Schluggi/AIOsense #7: Add simplified board

juvenalandres, 2022-11-23:
Hello,

This project looks great, thanks for sharing! 
I'm really interested in it but I was wondering if it is possible to simplify the pcb board, let me explain. I'm quite unexperienced in electronics and I found that soldering the BME280/680 sensor might be quite challenging, just soldering the pads on the bare sensor. 

This might be a lot of work but, could it be possible to make a version using [standard sensors](https://esphome.io/components/sensor/bme680.html), the same way the BH1750 is used? Or at least point me in the right direction to do it myself. 

Thank you

Schluggi, 2022-11-23:
Hey @juvenalandres,

We decided to go with SMD, because of the cost factor. But I can add some pins in v1.1 for these modules. So you can decide yourself. I also can't soldering these SMDs myself so i had to go to a public hacking space. I hope v1.1 is ready by end of these year :) 

Thank you for your feedback 😊

juvenalandres, 2022-11-23:
Thanks for the quick replay @Schluggi. 
I look forward to v1.1 then 😃

Schluggi, 2022-11-23:
@juvenalandres 
Just for you <3
[image]

commit 53a5c80b (2023-03-08): all-in-one-sensor is now AIOsense

added: additional I²C socket (closes #7)
added: logo & qr code (closes #5)
added: microphone support (closes #6)
added: schematic
changed: the BH1750 socket is now a generic I²C socket
fixed: PIR motion triggers onboard LED (fixes #1)
closed 2023-03-08 by Schluggi