Schluggi/AIOsense #9: Design a 3D printable case

Schluggi, 2022-11-24:


s-gordon, 2022-12-31:
Great work on the project all. I've actually designed a 3D-printable case for the board. Let me know if you're interested and I'd be happy to share.

MeisterGig, 2022-12-31:
Hey @s-gordon, 
we are also working on a Case for v2.0. I haven't posted it yet, since the PCB footprint is still not totally final. We are soon going to order an other batch of PCBs and hope to have a final version with case, to release.
In the meanwhile, here is a link to the WIP Case. https://cad.onshape.com/documents/0970230d187b540faa10c45f/w/4e628c030e78246416c5e857/e/a4e55bb6f45b9b87b321fa69?renderMode=0&uiState=63b00b6f2536af1c82a081f6

The case is supposed to have an assortment of mounting options available, which can be extended apon.

If you have a case for V1, please feel free to open a pull request so we can also offer an option for that. I am also open to improvement ideas :)

[image]

MeisterGig, 2022-12-31:
TODOs for the case until the release:
- [ ] Adjust to new footprint
- [ ] Make it thinner
- [ ] Get more light to the light sensor (might be solved by making the case thinner)
- [ ] Have a drawing for measurements for the mounting plate

cptbucky, 2023-03-20:
Excited for this, will get a 3d printer when it's ready to go 😁

cadriel, 2023-03-27:
It'd be great to see some further progress here;

- The lid doesn't appear to have a way  to attach to the case.
- The standing mount requires supports. It'd be better to have this printable without supports in some way.

MeisterGig, 2023-03-28:
> It'd be great to see some further progress here;
> 
> - The lid doesn't appear to have a way  to attach to the case.
> - The standing mount requires supports. It'd be better to have this printable without supports in some way.

The lid has ha notch which locks into place when pushing it in. And Yes the standing mount still needs support.

I have been refining the design after the official release of the V2.0 board. I'll follow up with further information soon.

afp316, 2023-03-30:
@Schluggi @MeisterGig Thank you for taking the time and effort to create this project.

I printed the WIP V2 case to trial fit.  Good news is that the board fit in the case perfectly.
[image]

I know that it's still in progress but just wanted to provide some feedback.  The lid doesn't have an interference fit so it won't stay in place without some kind of adhesive.  The corner wall mount should also be scaled a bit down to allow for a snap fit.

Did anyone else have issues or am I doing something wrong on the printing side of things?

Thanks!

MeisterGig, 2023-03-30:
Hey @afp316 ,
Thank you for your input :)  I also encountered problems with the lid. I inverted the notch on the lid and the case after the v2.0 rc1 because it was making it harder to get the board in and out of the case. I haven't  been able to get it as nicely  fit together as the previous version yet. Also i have resized the case recently and might have forgotten the mounts.

I'm gonna share some design considerations and problems i have encountered so far:
- The case should be modular
- It needs to have good ventilation for the bme both smd and recently also as a module (still working on that one )
- the light sensor needs to be as close to the lid as possible otherwise it won't get enough light or i need giant hole
- I personally like heat insert, but it was requested to not need any screws or heat inserts, because it looks bad and insert might be hard to get
- the PCB is very modular so it's hard to design a one size fits all case
-  the PCB should be easily taken out of the case
- when the case is closed it should fit securely inside the case
- pin headers have different hights (i have encountered 7, 8 an 8.5 mm versions)
- ESP8266 have there ports monted on the bottom esp32 on the top
- some people might want to supply power using the scre terminals below the mcu (i don't like a giant hole in the bottom)
- long pins on the lid to hold the board in place break off easily as to many screless locking mechanisms
- it should also look good

I was discussing with @Schluggi to maybe enforce one size of pin header and also maybe use heat inserts to mount the PCB. What are your thoughts on this :)

cadriel, 2023-03-30:
- Most of the pin headers i've encountered are 8.5mm. I don't think I've ever had anything else less than that.
- The more BOM items you add, the harder it is for people to source. Heat Inserts are a bit more difficult to find than screws. I've seen a bunch of cases that use screws as an option so it might be worth considering if you're having trouble. Remember - you can screw into plastic without the need of a heat insert, too.

IMO - It's going to be super hard to meet the needs of every single variation.

If you standardise the pin headers at 8.5mm (this seems the most common option) and start with supporting the recommended `S3` mcu - along with both the smd and pcb modules - then once you've sorted that, move on to variations. Perhaps you provide multiple case options for the different MCU's for example - rather than trying to create a "one case fits all" scenario.

Here's an example case that uses two self tapping screws. It uses little nubs at the bottom of the case to hold the PCB in place, and might provide some inspiration. :)
https://www.printables.com/model/380943-wled-wemos-shield-case-pcb-v3

afp316, 2023-03-30:
@MeisterGig I also like heat inserts but @cadriel makes an excellent point regarding sourcing and logistics.  I can confirm that I have the 8.5 mm headers measured with calipers.  Perhaps standardizing to a header spec would make the case design have one less variable (among many variables that you have dealt with).

As someone who doesn't really mess with SMD, I would think it's more accessible to design around the cheap breakouts that most amateurs buy (like myself).

I haven't looked at the datasheet for the light sensor or experimented with translucency effects on the sensor but perhaps a single layer/single wall would be able to satisfy the light sensor requirements for the application.  It may be such that the final recommendation is to print with certain colors of filaments for the lid.

In line with @cadriel thought, designing around a base set of recommended hardware will help with the design decisions immensely.  I'm sure you already know that considering I'm a late addition to this party. 🎉

cadriel, 2023-03-30:
Yep. Also - the light sensor doesn't necessarily need a whole bunch of light. I think its more important that the light exposure is consistent - and that you let SOME light in. I have other light sensor devices that only provide small holes to meet that requirement.

cadriel, 2023-03-31:
Oh - @MeisterGig  - the example case design I posted would help with some of your other concerns too. The base of the case is shallow - so adding / removing the PCB is very easy. The top of the case is inverted when compared to your current design, which provides other benefits too.

MeisterGig, 2023-04-11:
So I have adjusted a few details. The lid should now be able to snap onto the case again (if it needs too much force please tell me). There are many options variabilized (still a WIP), so adjustments can be made by changing the variables. At the moment I have set them to use 8.5 mm standoffs, a esp32 c3 mini and m3 heat set inserts (to change to screw, change the variable mounting_hole to fit your screw). The standing mount should now be printable without using supports. Unfortunatly I am currently unable to print  the case, so if someone else is able to do so feel free and give feedback. https://cad.onshape.com/documents/0970230d187b540faa10c45f/w/4e628c030e78246416c5e857/e/0f1cd3ab0062a4366c4b0a6d 

To address the light sensor issue. I have left the hole this big for now. Earlier testing back in v1 has shown, that readings from the sensor get to inaccurate, if there isn't enough light or even a grid or something like that in front of the sensor.

cadriel, 2023-04-12:
I'll print this today, but the first observation is that the overhangs on the mount are still problematic for FDM printing. Normally you'd want to keep these under 50° (or around-a-bout's there depending on the printer and material).

With the current mount, because you've used a fillet - the bottom edge is well over 50°. I've thrown in a screenshot here for reference;

[image]

If you go with a chamfer and control the degree's then it'd be an easier print.

Re: the other changes you made - they seem ok, I think it'll come down to the correct dimensions. I'll confirm with my setup - an ESP32 C3 and the Panasonic PIR and let you know if the table variables you have in place are good once I've printed and played around a bit.

cadriel, 2023-04-12:
[image]

The lid has something a bit odd going on.. =)

cadriel, 2023-04-13:
Ok, unfortunately there are multiple issues.
Including the above;

- The PCB now sits a bit higher, but the USB hole wasn't adjusted to match - and thus they don't line up.
- The mount won't fix to the case because the shape of the back interferes with the case itself.
- Even if the above wasn't true, the "clasp" wouldn't hold on to the case.
- The PIR won't seat into the PCB when the PCB is in the case, because the extrude doesn't sit deep enough (I.e., this cut needs to go deeper so the PIR can sit further into the case..).

Pics attached (in order to the points above);

[image]
[image]
[image]
[image]

Finally - here's an example when printing the standing mount.
[image]

I haven't printed the top yet due to the above. Do you have access to a printer @MeisterGig ? A lot of these issues could be caught with some iteration ;)

cadriel, 2023-04-13:
I just realized I have the mount around the wrong way, so ignore my comment about its interference. The cutout in the case is still irrelevant tho because the mount can't use it.

MeisterGig, 2023-04-13:
@cadriel Thanks for printing a test print. My calculations seem to be off after the variabilization of the cad files :( 

About the standing mount. My printer prints overhangs up to 70-80° quite fine, when it works... But maybe that's not the case for everyone. (And yes there are some degrees above 80°) i'll see how the campfer looks like.

cadriel, 2023-04-13:
> @cadriel Thanks for printing a test print. My calculations seem to be off after the variabilization of the cad files :(
>
> About the standing mount. My printer prints overhangs up to 70-80° quite [...]

Yep, I agree. I think it may also depend on material - but I think the key is catering for as many people and printers as you can. My Voron printing ABS has trouble with the first few layers - which then results in poor results further up on this part.

afp316, 2023-04-16:
- Did a test print in pla on my x1. Board slides in nicely.
- I haven't put the inserts in yet but the holes look pretty good by eye.
- The lid snaps in very snugly... almost too snugly. I couldn't remove it easily; had to use a screwdriver through the usb-c slot to pry.
- The mount clasps don't grasp the indentations on the case tightly enough. Could tighten up the clasp dimensions as-is or maybe implement slots in the case that slide onto and are held by the mount.
[image]

[image]

[image]

[image]

cadriel, 2023-04-16:
Is that the old case and stand with the new lid?

cadriel, 2023-04-16:
Oh, it looks like may @MeisterGig has adjusted it since I printed..

MeisterGig, 2023-04-16:
@afp316 the stand is not placed right. There are grooves in the middle wich it should go into.
It might be a little tall now i haven't tried it yet.

@cadriel yes i have been working on your feedback :)

MeisterGig, 2023-04-16:
@cadriel yes i have been working on it :) 

@afp316 thank you for the feedback.

MeisterGig, 2023-04-16:
@cadriel yes i have been working on it :) 

@afp316 thank you for the feedback. (I froze it as version v2rc2.2 in onshape)

Just got an idea on how to get rid of the inserts. @afp316 picture of the usb Port seeams to fit quite good now. I am gonna try to just use foam on the lid to keep the board in place.

My printer works again so I'm gonna print it now and tell you how it went by tomorrow :)

cadriel, 2023-04-16:
@MeisterGig Sounds good! On another note, you might want to adjust the base so that it sits lower. I found my USB-C cable couldn't be used because there wasn't enough space beneath the case when the stand was attached in its default position.

MeisterGig, 2023-04-16:
Do you mean the standing mount @cadriel ? I already made it longer and changeed it to a campfer like you suggested.

cadriel, 2023-04-16:
> Do you mean the standing mount @cadriel ? I already made it longer and changeed it to a campfer like you suggested.
> 
> You can already see it on the picture from @afp316

Yeah, the stand. But more about the distance between the bottom of the stand and bottom of the case. Here's a pic to illustrate - I ended up glueing the existing stand temporarily - but I had to glue it right at the bottom of the case otherwise I'd have no room for the USB to attach. Also - you can see its quite strained even in this position -  so having more distance here would be beneficial. I have other USB-C cables too - but the majority of them have long strain relief like the one in the picture.

I'm not sure if you've already made it taller or not - but yeah, it was an issue for me.

[image]

cadriel, 2023-04-16:
@MeisterGig also - re: holding the PCB in place, something like this might work?
[image]

MeisterGig, 2023-04-16:
@cadriel Yeah i had that idea aswell. But the pcb needs to be removable

cadriel, 2023-04-16:
> @cadriel Yeah i had that idea aswell. But the pcb needs to be removable

Aah, in what way? The PCB would still be removable in this case by first removing the top. Then it should fall out..

MeisterGig, 2023-04-16:
[image]

[image]

[image]

[image]

[image]

So i did some testing with the foam. It works quite well and allows for minor inaccuracies between different prints, pcbs,parts and is cheaper and easier to get then heat inserts or self tapping screws

afp316, 2023-04-17:
That is a great plan!  Well done!  It won't wiggle around or settle or anything like that, will it?  Even those that, for whatever reason, don't have the foam could just use a ball of electric tape to meet the same function as the foam.

For the HLK-LD2410, mine sits in the header firmly but can still wiggle around a little.  Is there anything that could be incorporated into the case that locks the free end of the HLK-LD2410 in place?  maybe a little standoff protruding from the lid?  Might be easiest to just put the same foam on the pcb under the mmwave pcb.

cadriel, 2023-04-17:
Hrm, is sandwiching the the processor against some foam going to cause overheating issues? I suspect the spec for the C3 may warrant specific spacing for airflow..

cadriel, 2023-04-18:
I've created a case and published it on printables.com
https://www.printables.com/model/456377-aiosense-case

This works well for me in the mean time! Feel free to use.

MeisterGig, 2023-04-18:
@Schluggi and I have done some Improvements tonight.

[image]

- The lid naps in easier @afp316. The mounts also fit better now (Cornermount, wall mount and camera Mount now fit very well)
- We moved the logo to the side which looks nicer.
- The LED now has a hole as well 

We think we can get rid of the foam all together by holding the back of the esp in place with a bar coming down from the lid. 

Also the standing mount doesn't print properly without support. Im gonna split it into two parts.

I am optimistic these improvements will be finished this week. The case will then released to the repo and thingiverse/printables.

afp316, 2023-04-18:
@cadriel Thanks for sharing your work!

@MeisterGig @Schluggi nice work all around.  I really like the changes you describe, especially the idea of the standoff coming off the lid to apply some pressure.

I can work on modeling a simple spacer to keep the mmwave sensor (the cheaper HLK-LD2410) from moving around freely.  Not sure if y'all are experiencing the same with your HLK's or maybe my tiny headers are junk.  I am using [these](https://www.ebay.com/itm/154730133714?var=454822447335).

MeisterGig, 2023-04-18:
@afp316 that would be great. I think i'm the only other person so far, who has an HLK for this project. I have soldered them directly to the board because I didn't bother to find pin headers yet. Of course it would be great to have them.

afp316, 2023-04-20:
I made an initial mount for the HLK-LD2410 to help a bit with keeping the pcb more secure.

[https://cad.onshape.com/documents/1a4aaa3e005976151241c26c/w/5e7ad8b8b7f30e1bc78c6254/e/c07d0c7cadc1cc9efcf0446a?renderMode=0&uiState=6441a2f1eb31156ab9b4474d](url)

[image]

I think I can do better than this so I'm going to try to use the thru holes for the SEN0395 to design a more compact and secure mount.

closed 2023-05-10 by Schluggi

riweston, 2023-05-13:
> @Schluggi and I have done some Improvements tonight.
>
> [image]
>
> * The lid naps in easier @afp316. The mounts also fit better now (Cornermount, wall mount and camera Mount now fit very well)
> * We [...]

Have you put together any additional case mount options aside from the corner mount? The case is great fit and I'm just trying to work out a good way to place it now and I don't really want to wall mount and have a cable trailing up the wall

Schluggi, 2023-05-13:
> Have you put together any additional case mount options aside from the corner mount? The case is great fit and I'm just trying to work out a good way to place it now and I don't really want to wall mount and have a cable trailing up the wall

Yes there is. But we are not happy yet. The clip for holding the case is to lose if you print this flat without support. So we need to adjust this in the future. But feel free testing.

[Mounts - Standingmount.zip](https://github.com/Schluggi/AIOsense/files/11469288/Mounts.-.Standingmount.zip)

riweston, 2023-05-13:
I modified the STL slightly and basically just copied the measurements for the clip er 'nubbins' from the existing corner bracket on printables, I'd say it could be a tighter fit but this seems pretty secure (like I can aggressively shake it without the aiosense falling out)

[Mounts - Standingmount_riweston.zip](https://github.com/Schluggi/AIOsense/files/11469681/Mounts.-.Standingmount_riweston.zip)

Schluggi, 2023-05-31:
@riweston I just printed your version of the stand and I have to say, I'm a big fan :)
Printable without support and a nice tight fit. 

With your consent I would love to share this on printables as part of the [official case](https://www.printables.com/de/model/477959-aiosense-case-official/files). I can link your github or printables profile as well. May I? If yes, I would love to have the onshape link to your version or a copy of the design as a file, since everything should be Open Source :)

Thank you!

riweston, 2023-05-31:
Oh great! Glad to hear it's a good fit :) 

I just moved a few vertices around on the existing design really but help to take credit 😂

Very happy to share the project file. I'll check the file size tomorrow and assuming it's not too big I'll raise a PR if that suits you?

commit 85d1efa2 (2023-05-10): added: 3d printable case (closes #9)