Schluggi/AIOsense #10: Add Support for Low Cost mmWave Sensor(HLK-LD2410)

MeisterGig, 2022-11-25:
I found a low cost mmWave Sensor on Ali Express, https://de.aliexpress.com/item/1005004786874722.html
[Datasheet](https://github.com/ESPresense/ESPresense/files/9189632/HLK-LD2410.user.manual.V1.02.pdf)
I haven't tried it yout yet, but if it works, it might be worth it to add pin headers for it, to being able to produce a low cost version of the AIOsense

Schluggi, 2022-11-25:
Sure...

Triskae, 2023-02-15:
I received 3 of theme today, I have to order a V2 board to try, @Schluggi do you have some boards home to test ?

Schluggi, 2023-02-15:
V2 boards yes. But no HLK-LD2410. Moreover is there no esphome config yet for these modules. But feel free and play around. Would be nice if you could create a PR if it's working for you :)

Triskae, 2023-02-15:
Can I email you about that ? I can send you a HLK-LD2410, or you can you send me a V2 board ? Instead of having to wait for it to be manufactured by JLCPCB ? :) I'll pay for the board and the ship costs of ! (I'm in france, we are almost neighbors xD

s-gordon, 2023-03-07:
@Schluggi @Triskae Some input from me here. I picked up a couple of HLK-LD2410s and had an earlier iteration of the V2 board to test these on (RC2, I think). I needed to use an adapter from the 0.05" pins of the LD2410 to the 0.1" female header on the board.

ESPHome support was added in [2023.02](https://esphome.io/changelog/2023.2.0.html) and the device page can be found [here](https://esphome.io/components/sensor/ld2410.html). After some tinkering with the config I have this working perfectly. I'll share pics and the config here when I have a sec.

koloodziej, 2023-03-07:
LD2410C has 0.1" pins but another pin layout. Can you make make PCB for that?
[image]

Schluggi, 2023-03-08:
> ESPHome support was added in [2023.02](https://esphome.io/changelog/2023.2.0.html) and the device page can be found [here](https://esphome.io/components/sensor/ld2410.html). After some tinkering with the config I have this working perfectly. I'll share pics and the config here when I have a sec.

@s-gordon
Awesome! Thank you for the update. I did not notice the ESPHome update yet. Feel free to create a PR to expand our ESPHome files :)

-----

> LD2410C has 0.1" pins but another pin layout. Can you make make PCB for that?

@koloodziej I'm sure we could implement this in v2.1. Could you create a separate issue for that? Links for buying also welcome.

closed 2023-03-08 by Schluggi

Triskae, 2023-03-13:
@s-gordon Great, that's some good news !! Thanks a lot for the update !

commit 6054780d (2022-11-25): added: pin header for HLK-LD2410 (closes #10)
fixed: labeling of first the Generic I²C