Schluggi/AIOsense #15: Evaluate v2.0-rc2

Schluggi, 2023-01-01:
We just bought some AIOsense v2.0-rc2.

They will arrive late January. 

Things we have to test:

- [ ] mmWave
- [ ] PIR
- [ ] BME280 (onboard)
- [ ] BME680 (onboard)
- [ ] onboard power
- [ ] IO pins

- CPUs
  - [ ] ESP32-C3
  - [ ] ESP32-S2
  - [ ] ESP32-D1
  - [ ] ESP8266-D1

- Modules
  - [ ] BME280
  - [ ] BME680
  - [ ] BH1750

Schluggi, 2023-01-11:
The PCBs arrived today. I'll test them within the next days.

Schluggi, 2023-01-13:
After testing a bit around i found out that the air quality stuff of the BME680 does not work at all with an non generic ESP32. 

Moreover there is a new problem with the PIR pin. If you want to use the PIR on the v2.0-rc2 board with an ESP8266 D1 mini, you have to remove the jumper and connect the center pin to GPIO13/MOSI. That's because GPIO0/D3 and GPIO2/D4 are always up and used for selecting the boot-mode.

s-gordon, 2023-01-18:
Thanks for the update @Schluggi! I was a little impatient and had some of the v2.0-rc2 boards fabricated as well. Can confirm that routing the PIR data pin to GPIO13 works for me as well.

What's the thinking here? Modify the design so that the jumper pin routes to GPIO13?

Schluggi, 2023-01-18:
> What's the thinking here? Modify the design so that the jumper pin routes to GPIO13?

We have notices some more problems with our current design:

- `bluetooth_proxy` only works with the arduino framework which is not compatible with non generic ESP32 boards ([source](https://esphome.github.io/bluetooth-proxies/))
- `web_server` and `captive_portal` only works with the arduino framework as well (would be nice to have)
- `bme680_bsec` are also not compatible. Without these module you cannot get values for IAQ, CO2 Equivalent & VOC Equivalent (temperature, humidity and pressure will work via the `bme680` module) 

I would drop the ESP32-C3/S2 and ESP8266 compatibility and go with the generic ESP32. But this would have major impact on the project. @MeisterGig is against it. 

We're happy to hear your opinion to this topic.

Schluggi, 2023-01-19:
After some more research from @MeisterGig he found out that `bluetooth_proxy` is actually working on the ESP32-C3 (since a big change in [ESPHome 2022.12](https://esphome.io/changelog/2022.12.0.html#bluetooth-proxy)).

He also found a [pull-request](https://github.com/esphome/esphome/pull/3500) for the `web_server` and `captive_portal` problem.

So `bme680_bsec` is the only module that's still not working. Maybe we can find a solution here as well.

florianmulatz, 2023-01-19:
> 680 does not work at all with an non generic ESP32.
>
> Moreover there is a new problem with the PIR pin. If you want to use the PIR on the v2.0-rc2 board with an ESP8266 D1 mini, you have to remove [...]

Do I understand correctly that a generic ESP32 (aka WEMOS D1 form-factor like this --> [ESP32 - AMZ Link](https://www.amazon.de/dp/B08BTYCGVV?ref=ppx_yo2ov_dt_b_product_details&th=1) will work without any problems including PIR and I2C Dev-Boards (like f.e. this one [URL](https://www.joy-it.net/de/products/SEN-BME680) )

Cheers

Schluggi, 2023-01-19:
> Do I understand correctly that a generic ESP32 (aka WEMOS D1 form-factor like this --> [ESP32 - AMZ Link](https://www.amazon.de/dp/B08BTYCGVV?ref=ppx_yo2ov_dt_b_product_details&th=1) will work without any problems including PIR and I2C Dev-Boards (like f.e. this one [URL](https://www.joy-it.net/de/products/SEN-BME680) )
> 
> If so - I highly appreciate to go for ESP32 (as mentioned above). It's anyhow the way to go for most projects (as of today :) )

On a software-side the generic ESP32 (aka WEMOS D1) works the best because we can use the slow, old but reliable arduino-framework inside ESPHome. There is a newer and faster framework called "esp-idf". But unfortunately not all modules are supported right now. 

The WEMOS D1 generic ESP32 Board has a different pin layout then the ESP8266, ESP32-C3 and ESP-S2, so it will not work with any version of AIOsense yet. 

The I²C dev-boards will work with every version of AIOsense but depending of the ESP the feature set of the BME680 is different. If you want to get all out of your BME680 use an ESP8266 and the v2.0-rc3 release. But because its a ESP8266, it has no bluetooth. The PIR will work with [this](https://github.com/Schluggi/AIOsense/issues/15#issuecomment-1382575615) modification on every board v2.0-rc2 and above. 

The ESP32-C3 has some benefits over the generic ESP32 (USB Type-C, no programmer needed). We plan also to release a SMD only version of AIOsense. So we should have this in mind. No need for a programmer chip is very nice and cuts down the costs.

So i opened an [ESPHome feature request](https://github.com/esphome/feature-requests/issues/2072) to support the ESP32-C3. This would be the best solution in my opinion. If there is no progress until the end of January, we will switch completely to the generic ESP32.

florianmulatz, 2023-01-20:
> The WEMOS D1 generic ESP32 Board has a different pin layout then the ESP8266, ....

I disagree :) From my point of perspective they're pin-compatible (I can not say anything about ESP-C3 / ESP-S2 - since I don't konw (neither own) them)

[image]

Please just let me know if I get this wrong :)

s-gordon, 2023-01-20:
> > The WEMOS D1 generic ESP32 Board has a different pin layout then the ESP8266, ....
> 
> I disagree :) From my point of perspective they're pin-compatible (I can not say anything about ESP-C3 / ESP-S2 - since I don't konw (neither own) them)

Agree with @florianmulatz. I've tried this with a generic ESP32 (MH-ET Live MiniKit) and it seems fine. I've only tested THT BME280, BH1750, SEN0395, and the PIR module. Happy to lodge a PR with my YAML if there's interest.

Schluggi, 2023-01-21:
> Agree with @florianmulatz. I've tried this with a generic ESP32 (MH-ET Live MiniKit) and it seems fine. I've only tested THT BME280, BH1750, SEN0395, and the PIR module. Happy to lodge a PR with my [...]

I totally agree. I wonder how i could miss that. Can you upload an image of the ESP32 on the board? PRs are always welcome :)

I just ordered a few ESP32 mini boards to test it myself. 

Moreover I changed the labels of the ports. Now you can see more easily whats connected to. That should help by finding the right port in the ESPHome-config. Any comments on that?

[image]

Next steps:
I will test the ESP32 D1 mini board. If everything works fine I will finally release v2.0. Thank you for your help :)

s-gordon, 2023-01-22:
Sure. These aren't the best pictures, but here you go :-). I salvaged the ESP32 board from another project and needed to remove the old header pins - hence why the board looks a little worn. I'm waiting on the PIR sensors to arrive later this week, and so they aren't shown here.

No worries. I'll look into doing a PR later in the week.

[image]
[image]

Schluggi, 2023-01-22:
@s-gordon 

Thanks! :)
Do I see it right that the ESP32 mini is bigger than the AIOsense board?

s-gordon, 2023-01-22:
That's right, it does protrude beyond the bottom edge of the AIOsense board by about 3 mm. The camera perspective exaggerates it a little.

Schluggi, 2023-01-30:
I finally found a way to use the `bme680_bsec` component with an ESP32-C3 (fb7976eb35cdcf3bc050d74ac4c29ba4ce0aaad7). 🥳 

Everything works now!

So the only thing we have to discuss: Does we want to increase the board size to fit better with generic ESP32 boards? I would say no, because otherwise we would have a problem with the case.

But maybe @MeisterGig could come up with a case design option for generic ESP32 boards? So we don't have to increase the board size, just the case?

s-gordon, 2023-01-31:
I'm in favour of increasing the PCB size to accommodate generic ESP32 boards, although I am being a little selfish :-P.

If reworking the case design is a concern, then I'd be happy to contribute the STLs that I've put together. This will fit both the D1 Mini and generic ESP32.

MeisterGig, 2023-02-01:
I agree with @Schluggi. It is unwise to increase the size of the board even more. It's already quite big. As far as i can see the length would also need to be modified, this would be a disadvantage for c3 and s2 modules, because the USB port would be too far in the case.

In addition, c3 and s2 chips are newer, cheaper and will be supported for a longer time compared to the 32 series https://www.espressif.com/en/products/longevity-commitment

MeisterGig, 2023-02-01:
I agree with @Schluggi. It is unwise to increase the size of the board even more. It's already quite big. As far as i can see the length would also need to be modified, this would be a disadvantage for c3 and s2 modules, because the USB port would be too far in the case.

In addition, c3 and s2 chips are newer, cheaper and will be supported for a longer time compared to the 32 series https://www.espressif.com/en/products/longevity-commitment

If we are looking a little bit further in the future. Espressif just launched the ESP32-C6 (which has WiFi 6 and Zigbee support) https://www.espressif.com/en/products/socs/esp32-c6 this chip also has a similar size as the c3 and comes with an internal USB to Serial module like the c3 and S2. So my best guess is, that WEMOS will probably design the c6-mini in the same form factor as the c3 and s2 mini.

florianmulatz, 2023-02-02:
Agree to. Although I've more esp32 (d1mini) at home - I'm willing to invest in a more modern platform as well of course. ESP32-c6 sounds cool - ZIGBEE let's my heart beats faster 😅

Schluggi, 2023-02-02:
> Although I've more esp32 (d1mini) at home - I'm willing to invest in a more modern platform as well of course. 

You don't have to. The generic ESP32 will be compatible. We just don't increase the board size for this one. So it overlaps as @s-gordon pictures has shown. @MeisterGig will also release a case design for this one. So you can still use your generic ones. However the C3 will fit better.

Yes, the C6 looks cool but please have in mind we have to wait until ESPHome and Bosch supports it. 

At this moment I will release v2.0-rc4 in the next days and order these. Afterwards we hopefully have a bug free v2 release :)

closed 2023-02-05 by Schluggi