Schluggi/AIOsense #18: Request for enhancement (RFE): Redesign the 5V Power Input

florianmulatz, 2023-01-19:
Hi Schluggi,

I suggest to redesign the power input to not use a "Male PIN-Header" but a "wire-to-board-terminal-block" like this one --> [WUERTH Terminal-block](https://at.farnell.com/wurth-elektronik/691216410002/reihenklemme-wtb-2way-30-12awg/dp/3768503)

At least on the ESP32, the USB port is on the top. I'm not sure if the ESP8266 has it on the bottom, like your 3D model. Since you won't be supporting the 8266 anymore anyway, it would at least be worth considering, because with a terminal block you might have more strain relief

[image]

Cheers Flo

Schluggi, 2023-01-19:
@florianmulatz, thank you for your input. I'll keep this in mind before the final v2 release. It is something we can quickly change. 👍

Schluggi, 2023-01-19:
I just tested the tolerances with my v2.0-rc2 board with an ESP8266. Its really tight! 😄 
So maybe we can move the IO-ports under the ESP and the power connector to the IO-panel. We'll see :)

florianmulatz, 2023-01-20:
Does the "tightness" makes it not good? I think there will be terminal-blocks with even a smaller "housing" at least this one was something I had at home. Just for reference (it has a height of 9,95mm) - I don't know the exact manufacturer of this one - since I bought it very cheap on AliExpress.

The one from Farnell above is wrong - I did not take care on the pitch. This one should fit better --> [WUERTH Terminal-Block](https://www.digikey.at/de/products/detail/w%C3%BCrth-elektronik/691210910002/10668428). It has a height of only 8,5mm

Cheers

Schluggi, 2023-01-21:
8,5mm sounds great. I changed it :)

[image]

florianmulatz, 2023-01-23:
Thx for your fast update! - Issue closed

commit 0327d0da (2023-01-21): changed: power pins replaced with terminal block (closes #18)
closed 2023-01-23 by florianmulatz