Schluggi/AIOsense #23: Issues with mmwave GPIO pin

MeisterGig, 2023-02-08:
I am currently having issues with the mmwave GPIO pin. Sometimes when the pin i high, it isn't detected by the ESP. I checked the connection, which seems to be okay. Maybe the issue is the internal pulldown.

Anyways I was searching for other solutions. I found out that the mmwave sensor is also sending the value over UART so i tried to write a custom binary sensor component for that, but haven't gotten far due to my horrible C skills.

But i now also found that someone else might have already done the work:  https://github.com/esphome/esphome/pull/4203

We could evaluate this solution and maybe save a gpio pin, which can then be used for something else, like audio output.

Schluggi, 2023-02-13:
Does the any other ESP work for you?

Schluggi, 2023-03-08:
@MeisterGig any update? Otherwise I would close this issue.

MeisterGig, 2023-03-08:
I haven't tried it anymore.

Schluggi, 2023-03-28:
I'll close the issue. You can reopen it in case of new information.

closed 2023-03-28 by Schluggi