Schluggi/AIOsense #24: PIR not responding on version 1

cptbucky, 2023-02-11:
First of all thank you for making this project available, it's exactly what I was looking for. 

I got a bit excited and ordered the version 1 pcb without reading the warning, is there somewhere that lists out the things that don't work on v1?

To the point, when testing the EKMC1603111 PIR it's not responding to motion when attached to the v1 pcb. I have 2 test pcbs, one is fully soldered with all components/female headers, the other just the dev board and PIR socket (assuming that's the bare minimum requirements).

If I attach the sensor directly with a nodeMCU or Wemos D1 mini ESP32 it works perfectly, attached to an Olimex POE ISO I have to use a pull down resistor from ground to out. I have still not been able to get the Wemos D1 mini ESP8266 to work, which behaves the same as when connected to the pcb.

Any help would be amazing!

Schluggi, 2023-02-11:
Hi Tom,
thank you for issue. Issue #13 is related to this. Sure, without any info you couldn't know.

What does not work is the PIR (wrong footprint) and maybe also the onboard BME sensors (if your power supply has voltage drops).

To fix your PIR you can cut the PCB paths with a cutter and resolder a wire. Please take a look into the v2 PIR footprint where's fixed.

The Esp8266 does not work with the v1 board as well, because the PIR pin is connected to a pin which is not allowed to be pulled up at bootime.
In general the PIR does not need a pulldown or pulluo resistor.

I would recommend the ESP32-C3 mini for all further boards. The documentation is wrong at this point and still recommend the generic ESP32, which works but is a bit to big for the PCB v2 (never tested with v1).

I hope I could help :)

cptbucky, 2023-02-12:
Thanks ever so much for the quick reply @Schluggi it's all very helpful so thank you for that!

Am I correct that the wrong footprint for the PIR on v1 is VDD and GND being the wrong way round? If so, forgive my inexperience here but assuming I use the ESP32 board I can cut the track for VDD 3v but I'm not sure how to intercept GND do you mean that I should disconnect it from the board itself and use a wire instead?

Fortunately I have a few ESP32-C3s on order so looking forward to v2!

Schluggi, 2023-02-13:
Your welcome :)

Yes your are right. VCC/VDD and GND are swapped. Since we use a hole layer as GND its harder to intercept. All GND pins has 4 lines to the layer you have to cut them all (red=cut):
[image]

But yes. Waiting for V2 or use only wires to connect the PIR would be much easier.

---

V1 (wrong): 
[image]

v2 (correct):
[image]

cptbucky, 2023-02-14:
Thank you for the excellent explanation, very much appreciated!

closed 2023-02-16 by cptbucky