Schluggi/AIOsense #43: C3 Configuration Yaml not working

florianmulatz, 2023-03-15:
**Category**
- [ ] Hardware / PCB
- [X] Software / ESPHome

Error during config-validation in ESPHome (looks like the "external component" does not get loaded ?!
ESPHome-Version: | v2023.2.4 

```
INFO Reading configuration /config/esphome/esphome-mmwave-schlafzimmer.yaml...
Failed config

sensor.bme680_bsec: [source /config/esphome/esphome-mmwave-schlafzimmer.yaml:141]
  
  Component sensor.bme680_bsec requires component bme680_bsec.
  platform: bme680_bsec
  temperature: 
    name: aiosense-c3-mini Temperature
    sample_rate: lp
    filters: 
      - median
  pressure: 
    name: aiosense-c3-mini Pressure
    sample_rate: lp
    filters: 
      - median
  humidity: 
    name: aiosense-c3-mini Humidity
    sample_rate: lp
    filters: 
      - median
  iaq: 
    name: aiosense-c3-mini IAQ
    id: iaq
    filters: 
      - median
  co2_equivalent: 
    name: aiosense-c3-mini CO2 Equivalent
    filters: 
      - median
  breath_voc_equivalent: 
    name: aiosense-c3-mini Breath VOC Equivalent
    filters: 
      - median
text_sensor.bme680_bsec: [source /config/esphome/esphome-mmwave-schlafzimmer.yaml:172]
  
  Component text_sensor.bme680_bsec requires component bme680_bsec.
  platform: bme680_bsec
  iaq_accuracy: 
    name: aiosense-c3-mini IAQ Accuracy
```

florianmulatz, 2023-03-15:
if I put 
```
bme680_bsec:  # disable for bme280
  temperature_offset: "5.8"
  state_save_interval: "1h"
```
into yaml (from 8266 configuration) - Validation works again

Schluggi, 2023-03-15:
Thanks for noticing. I will fix it later this day. Your workaround is actually the fix. It does not work without `bme680_bsec:`.

commit f2393915 (2023-03-15): added missing bsec config (fixes #43)
closed 2023-03-15 by Schluggi