Schluggi/AIOsense #44: resistors R1/R2

florianmulatz, 2023-03-16:
**Category**
- [ ] Some docs are wrong or outdated
- [X] I miss something the docs

There should be an info that the two resistors R1/R2 must not be populated if using I²C module for using BME(280|680). I got quite bad readings because I soldered them initially to the board - sometimes no values got populated at all during start - I had to do several restarts.

Maybe I'm missing something or misinterpreting something - but since I cut the two resistors - everything is doing fine.

Cheers

Schluggi, 2023-03-16:
These resistors are necessary for I²C in general. They pull-up the signal to define the I²C master.

Which ESP do you use? Could you send a picture of your module where I can see the onboard resistors on the module?

florianmulatz, 2023-03-16:
I'm using the new C3 Lolin ESP32. BME680 is exactly this module --> https://de.aliexpress.com/item/1005001621661986.html?spm=a2g0o.order_list.order_list_main.5.21ef5c5fHqQz76&gatewayAdapt=glo2deu

Schluggi, 2023-03-18:
You might be right. I will analyze this in detail in a few days.

cptbucky, 2023-03-18:
I have had an odd behaviour that might be related whilst using the C3 Lolin ESP32, the PIR is unresponsive, always reporting detected.

Thanks for the above, I will try removing the resistors as I have 2 AIOSense boards setup with consistent behaviour (no I²C issues though!) and one custom D1 Mini ESP8266 that works. I haven't added the surface mount capacitor.

If entirely unrelated, please let me know and I'll raise a new issue.

s-gordon, 2023-03-19:
I've seen this behaviour with ESP32 D1 Minis and Lolin C3 Minis and the BME280 when fixed to the board. Typically failed I2C scans, or odd errors in the log. Multiple restarts and re-installing firmware eventually fixed it. I'd put this down to bad soldering as I had done this using solder paste and a hot air gun, but it may be similar to what others have reported here.

Schluggi, 2023-03-20:
We need I²C pull-up resistors to make I²C to work. But since every module has there own pull-ups, we ending up with too many pull-ups which can resolve in bad readings. 

I spotted pull-ups on your BME-module as well as on the light module. So yes you're right you can remove  the onboard resistors as long as don't use the SMD-BME module only.

Maybe it would be nice to remove the SMD parts completely. That's why I'm started the poll: https://github.com/Schluggi/AIOsense/discussions/51.

Schluggi, 2023-03-20:
@florianmulatz I just added this to the docs.

Again: Thank you for noticing :)

closed 2023-03-20 by Schluggi