Schluggi/AIOsense #48: BOM + Soldering / Part instructions

cadriel, 2023-03-19:
- [* ] missing something the docs

Would love to see some further docs re;
- A BOM. Could simply be a page that links to recommended parts on DigiKey and Mouser perhaps. This could help with confidence for some people.
-  Some instruction / imagery that represents what a populated PCB looks like. This also helps to confirm visually that you've soldered your board correctly.
- A guide on how to test your board is soldered correctly with a multimeter? I'm not sure if this step is necessary given the simplicity of the PCB.

Also, i wanted to ask about PIR sensors and dirty power. I get a lot of false positives with my Panasonic PIR that I intend to use on this project - and read that its often a good idea to do something to avoid this. I wasn't sure if this is something that's covered in this project yet or not (or even if its required..).

Finally, THANK YOU for this! I can replace my hobbled together sensor with a decent PCB, finally. :0

cadriel, 2023-03-20:
I've since found most links in the docs, so ignore my first comment. The mouser link for the TO-5 is / was down when I was looking. Might be nice to add the equivalent DigiKey link there (https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/917-43-103-41-005000/1212170?s=N4IgTCBcDaIJwEYDsBaALAZhQgDFtCKOOArMTiALoC%2BQA)

Schluggi, 2023-03-20:
Hi @cadriel,

thank you for your suggestions.

----

> Might be nice to add the equivalent DigiKey link there

Yes. The Mouser website is often in maintenance mode. At least at weekend. I just updated the docs: https://github.com/Schluggi/AIOsense/commit/2e7f5f7586525528304d58c9393b6f659e8f3144 :)

----

> A BOM. Could simply be a page that links to recommended parts on DigiKey and Mouser perhaps. This could help with confidence for some people.

I have often thought about a complete BOM but since everything is modular it felt hard. Do you thing it would be more beginners-friendly to have one?

----

> Some instruction / imagery that represents what a populated PCB looks like. This also helps to confirm visually that you've soldered your board correctly.

Did you spot the images at the [README.md](https://github.com/Schluggi/AIOsense/blob/main/README.md#what-does-it-look-like-)? Or are you looking for some closer images?

----

> A guide on how to test your board is soldered correctly with a multimeter? I'm not sure if this step is necessary given the simplicity of the PCB.

This would make sense in the future when the PCB is more complex but at this moment it is (as you said) a really simple PCB.

cadriel, 2023-03-20:
> Hi @cadriel,
> 
> thank you for your suggestions.
> 
> > Might be nice to add the equivalent DigiKey link there
> 
> Yes. The Mouser website is often in maintenance mode. At least at weekend. I just updated the docs: [2e7f5f7](https://github.com/Schluggi/AIOsense/commit/2e7f5f7586525528304d58c9393b6f659e8f3144) :)

Fantastic! :)

> 
> > A BOM. Could simply be a page that links to recommended parts on DigiKey and Mouser perhaps. This could help with confidence for some people.
> 
> I have often thought about a complete BOM but since everything is modular it felt hard. Do you thing it would be more beginners-friendly to have one?

I think so yes. I've completed a bunch of other similar projects, most notably a WLED wemos shield recently. I found their BOM quite useful. Especially given the fact that many users are from many parts of the world - so having multiple links for major regions is quite helpful. Also, having a single place to reference all parts is also helpful.

**Note:** I could not find links to the resistors at all in the documentation - so I know that should **definitely** be resolved in some way.

Here's the project for reference; https://github.com/srg74/WLED-wemos-shield/wiki

> Did you spot the images at the [README.md](https://github.com/Schluggi/AIOsense/blob/main/README.md#what-does-it-look-like-)? Or are you looking for some closer images?

Yes - I noticed the images on the main README. It might be helpful to see a couple of configurations here. I.e., some pictures of someone with a configured board with only the TO-5 and resistors in place (so people can confirm orientation) - and then a couple of pictures that represent the SMD versus the PCB modules installed.

> > A guide on how to test your board is soldered correctly with a multimeter? I'm not sure if this step is necessary given the simplicity of the PCB.
> 
> This would make sense in the future when the PCB is more complex but at this moment it is (as you said) a really simple PCB.

Yep. Here's the WLED projects wiki page on the subject which might be useful for reference; 
https://github.com/srg74/WLED-wemos-shield/wiki/Assembly
https://github.com/srg74/WLED-wemos-shield/wiki/Check

It really helps people that are less familiar with electronics. :)

> 
> > Also, i wanted to ask about PIR sensors and dirty power. I get a lot of false positives with my Panasonic PIR that I intend to use on this project - and read that its often a good idea to do something to avoid this. I wasn't sure if this is something that's covered in this project yet or not (or even if its required..).
> 
> Which CPU are you using? Are you using the EKMC1603111? Which board version are you using? Does the PIR work in general or is it flipping all the time? I use the PIR since a couple of months without any problems.

My current configuration is a hand soldered version using the commonly known ESP32 D1 Mini
https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html
https://www.amazon.com/gp/product/B07BK435ZW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

And the PIR sensor is the Panasonic EKMC.
https://www.digikey.com/en/products/detail/panasonic-electric-works/EKMC1603113/2601882

The PIR is connected to pin 17 I believe. Its been getting worse over time, and I think its due to dirty power - but I'm no expert on the subject. If I disconnect power to the device, then it's good but slowly returns to sending false positives every few minutes.

There are a few posts on the home assistant forums from people more knowledgable than me suggesting that power can cause issues with PIR's. I'd guess the best bet is to follow the datasheet for the EKMC tho.
https://community.home-assistant.io/t/which-pir-sensor/512976

https://api.pim.na.industrial.panasonic.com/file_stream/main/fileversion/248121
https://api.pim.na.industrial.panasonic.com/file_stream/main/fileversion/244673

It could also be poor wiring on my part. I've ordered your RC4 PCB today, so I'll be connecting it all in the coming weeks to see if that resolves my false positive issues.

Sorry for the wall of text, hopefully you can take that all in :)

cadriel, 2023-03-29:
You're awesome, thanks.

My PIR issues were for a manually soldered setup, nothing to do with this project. I'll re-test this in the new configuration with your PCB today to see if I have similar issues or not.

afp316, 2023-03-30:
I purchased these BME680 breakouts from [aliexpress](https://www.aliexpress.us/item/2255800927135555.html?spm=a2g0o.order_list.order_list_main.23.719d1802knSJkp&gatewayAdapt=glo2usa&_randl_shipto=US).

I started assembling the board (V2.0.0) with headers and noticed that the BME680 module pins are oriented such that the breakout will extend away from the center of the board when installed.  See the image below.

[image]

Was this by design?  Maybe I bought the wrong ones.

Schluggi, 2023-03-31:
> Was this by design? Maybe I bought the wrong ones.

Please take a look at #55. For now you have to solder it upside down. This is already fixed and will be part of v2.1.0.

afp316, 2023-03-31:
I guess I didn't look hard enough.  Thanks!

afp316, 2023-04-15:
I had some stacking headers lying around so I used them with the V2.0 pcb.  I bent the header at a right angle to the board.  This allows for avoidance of unsoldering/resoldering the header for the bme680 pcb.  With this, the module protrudes upward and I'll probably just modify the case a bit for this use since the next version has a modified pcb.
[image]
[image]

Schluggi, 2023-04-23:
@cadriel We now have a BOM. Please give feedback :)

cadriel, 2023-04-23:
@Schluggi The link in the docs appears to be broken. It takes me here;
`https://github.com/Schluggi/AIOsense/bom` which is a 404.

Maybe it was meant to link here `https://github.com/Schluggi/AIOsense/tree/main/bom`

Schluggi, 2023-04-23:
> Maybe it was meant to link here `https://github.com/Schluggi/AIOsense/tree/main/bom`

Thanks. Is fixed now :)

cadriel, 2023-04-23:
Also, the very first next button on the docs homepage takes you here;
https://aiosense.readthedocs.io/en/latest/_pcb/

But this page doesn't seem to be in the menu anywhere, and it looks like this is the only way to get to it. However, it doesn't seem like the first logical place you end up on? Might've been a weird oversight or something.

The page itself seems relevant tho - so maybe it just needs to be put in the menu somewhere?

cadriel, 2023-04-23:
Re: the BOM - it might be worthwhile to have this as a google docs (or similar) link instead? This way you can update it on the fly without having to generate a PDF, or make a commit.

You can then also start providing links to components more easily, which might also be a good improvement.

For example, here's the voron sourcing guide.
https://docs.google.com/spreadsheets/d/e/2PACX-1vS_jv9UtoSkgQoYD47VE9wIpTnjsEbzAREPG8895b6UL0v99OIfVEJQ4BHl2w_Rn6V0YUe1ldfPjone/pubhtml?widget=true&amp;headers=false%22%3E#gid=

cadriel, 2023-04-23:
(But it otherwise looks good for now too! :))

Schluggi, 2023-04-23:
> Also, the very first next button on the docs homepage takes you here; https://aiosense.readthedocs.io/en/latest/_pcb/

Fixed. I thought an underscore would make the site inaccessible :/

--
Google docs cloud be an option, but I like to have everything in repos. And since the BOM will not change that often...