Schluggi/AIOsense #52: PIR Sensor Unresponsive

cptbucky, 2023-03-20:
**Category**
- [x] Hardware / PCB

**Describe the bug**
Apologies this doesn't appear straightforward!

When booting the board with a Panasonic PIR attached (EKMC1603111) the reading for Motion in Home Assistant is constantly Detected.

Initially when setting up the boarding everything was fine for about a week, then the behaviour started so I replaced the PIR which fixed the issue; around a week later the same issue occurred :-( I question whether this was actually triggered by an outlet change.

The testing setup I now have: 

1. AIOSense board setup with all components, including resistors on a ESP32 Lolin C3 Mini  - reports Detected
2. AIOSense board setup with only the PIR, no resistors on an ESP32 D1 Mini - works, though only running for a couple of days
3. custom board with all components, no resistors on an ESP8266 D1 Mini - works, has been running for a while and hasn't faced this issue yet
4. breadboard mimicking the schematic with all components, no resistors on a ESP32 Lolin C3 Mini - works, though only running for a couple of days

If I switch the ESP32 boards over on 1 and 2 then both report detected, switching them back and it appears scenario 2 has now broken :face_exhaling: 

In the above scenario 1-3 are on a shared extension brick whilst 4 is on a separate outlet, the reason I mention is that the breadboard also displayed this behaviour but only on the shared extension brick and not when connected to my laptop or the separate outlet.

If anyone can think of anything, wants me to try anything then by all means let me know.

sebastiansson, 2023-03-29:
I had to set this in order for it not to stay on "Detected" all the time:

```yaml
pin: 
  number: GPIO23
  mode: INPUT_PULLDOWN
```

Schluggi, 2023-03-29:
> I had to set this in order for it not to stay on "Detected" all the time:

There is no hint for a PULLDOWN in the datasheet for the PIR. And I don't even need one. But because this had no negative impact I'll change it in the example.

@cptbucky can you test if that solves you PIR problem?

sebastiansson, 2023-03-30:
If I recall correctly it worked without `INPUT_PULLDOWN` when I had it connected to my computer and flashed it. But when I plugged it in the wall without reflashing it, the sensor stayed on "Detected" all the time. Maybe the`INPUT_PULLDOWN` resets the sensor somehow?

Schluggi, 2023-03-30:
Ok, please try the following: Disconnect the PIR, plug the board into your wall, wait until the board is online and connect the PIR again. 

Post your result :)

sebastiansson, 2023-03-30:
> Ok, please try the following: Disconnect the PIR, plug the board into your wall, wait until the board is online and connect the PIR again.
> 
> Post your result :)

Without `INPUT_PULLDOWN` I suppose?

Schluggi, 2023-03-30:
> Without `INPUT_PULLDOWN` I suppose?

Yes

cptbucky, 2023-04-04:
Following on from the `INPUT_PULLDOWN` configuration addition I have not experienced the always detected behaviour.

I haven't removed it and rejigged to try and force the behaviour again, I will try to do a bit more testing on this.

cptbucky, 2023-04-14:
I've had a play and not seen the issue recur following the addition of the pull down config so I think this issue can be closed.

Thanks for everyone's help on this.

closed 2023-04-14 by cptbucky