Schluggi/AIOsense #53: PIR / SEN0395 does not fit together

florianmulatz, 2023-03-22:
**Category**
- [X] Hardware / PCB
- [ ] Software / ESPHome

**v2 - RC4**
...

It looks like that the PIR does not fix together with SEN0395 (at least when mounted via "socket")
Did you guys managed to get this together?
...
[image]
[image]

Schluggi, 2023-03-22:
Hi @florianmulatz,

you need a TO-5 3-pin socket: [Mouser](https://www.mouser.de/ProductDetail/Mill-Max/917-43-103-41-005000?qs=teNCaa%2FZ3auQVxVMs%2F5ihg%3D%3D), [Digi-Key](https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/917-43-103-41-005000/1212170?s=N4IgTCBcDaIJwEYDsBaALAZhQgDFtCKOOArMTiALoC%2BQA)

This is noticed in the docs [here](https://aiosense.readthedocs.io/en/latest/sensors/#motion-sensor-pir).

closed 2023-03-22 by Schluggi

florianmulatz, 2023-03-22:
ok thx - didn't notice that!

florianmulatz, 2023-03-22:
I found those sockets too expensive to be honest  so I cheated :) 
It's not "beautiful" but it works and when it's in it's case afterwards nobody would recognize :)

Cheers
[image]