Schluggi/AIOsense #57: PIR interferes with mmWave

cadriel, 2023-03-29:
**Category**
- [*] Hardware / PCB

**Version of ESPHome / AIOsense**
2.0 RC-4

**Describe the bug**
The PIR sensor interferes with the mmWave sensor. Accordingly - the PIR doesn't fully seat into the socket.

[image]

The socket is fully seated, it's designed to sit a couple of mm above the board, as per the spec here. Note this is the spec'd TO-5 socket from the documentation.

[image]

The PIR sensor itself can't be fully seated due to its pin length (you could of course cut these down if required..) however, in its default state the PIR sensor interferes with the soldered pins on the DFRobot mmWave sensor.

Secondarily, the current printable case doesn't work at all with this configuration because its cutout sits far too high to even seat into the socket at all. I assume the case would have an update to resolve this we haven't seen yet.

You'd either need a TO-5 socket that sits higher, or would need to adjust the position of the PIR sensor to resolve this.

Schluggi, 2023-03-29:
The PIR height is really precise calculated and it's made for exactly this height. However I was surprised as well the first time I had put the PIR into the socket. But even if this is not beautiful, it works fine. Moreover I was not able to find any other (cheap) TO-5 Socket. If you spot one please share it.

Did you print the case from the files (it's a case for v1 only). Or did you use the beta version of [ this case](https://github.com/Schluggi/AIOsense/issues/9)? However there is no perfect case yet. Please notice #9. To prevent further confusion, I've moved the case file into a subdirectory named by the version.

Schluggi, 2023-03-29:
> The case I printed was the work in progress for V2.. not V1.

The case need a complete rework @MeisterGig is working on this.

closed 2023-03-29 by Schluggi