Schluggi/AIOsense #59: add aliexpress link for TO-5 3-pin socket

regevbr, 2023-03-29:
**Category**
- [x] Hardware / PCB
- [ ] Software / ESPHome

**Whats your idea?**
I want to order `TO-5 3-pin socket` but the shipping is extremely expensive. I tried to look for it in aliexpress, but without luck. Can you please help out?

For EKMC1603111 i managed to find https://www.aliexpress.com/item/1005003203976549.html?spm=a2g0o.cart.0.0.7d1f38dapwtcSv&mp=1 but I can't verify it is reliable yet

Schluggi, 2023-03-31:
Hi @regevbr,

I have not found the socket there either.

Maybe [Octoparts](https://octopart.com/917-93-103-41-005000-mill-max-853810?r=sp) can help to find a cheap solution.

regevbr, 2023-04-01:
> Hi @regevbr,
> 
> I have not found the socket there either.
> 
> Maybe [Octoparts](https://octopart.com/917-93-103-41-005000-mill-max-853810?r=sp) can help to find a cheap solution.

Thanks sadly shipping to Israel it is around 28 USD in all oh the websites :-(

Schluggi, 2023-04-03:
If you can't get a socket. You could go [the dirty way](https://github.com/Schluggi/AIOsense/issues/53#issuecomment-1479398825).

I close the issue for now. If you can confirm that the PIR from Aliexpress works, please reopen this issue or (even better) create a PR to extend the docs. 

Thank you for you contribution.

closed 2023-04-03 by Schluggi