Schluggi/AIOsense #60: BME680 erro

afp316, 2023-03-30:
**Category**
- [ ] Hardware / PCB
- [x] Software / ESPHome

**Version of ESPHome / AIOsense**
ESPHome: 2023.3.2
AIOsense: latest

**Describe the bug**
I get an error with the bme680 initialization.  It wants me to append "_m" to several variables in the cpp file.  I searched to find this error but couldn't come up with anything.  I apologize if there is already a solution.  See the link for my logs.

[logs_aio-utility-room_run.txt](https://github.com/Schluggi/AIOsense/files/11107146/logs_aio-utility-room_run.txt)

Schluggi, 2023-03-31:
Hi @afp316,

please share your YAML as well :)

afp316, 2023-03-31:
oops crap, sorry about that.  noob moment.  Thanks for you patience..

[aio-utility-room.yaml.txt](https://github.com/Schluggi/AIOsense/files/11122228/aio-utility-room.yaml.txt)

I haven't included any pw's or keys in the txt.

Schluggi, 2023-04-04:
Thank you @cadriel for the quick fix.

Unfortunately if you want to use bsec you have to downgrade your ESPHome. 

It should be possible to use the new bsec component @cadriel had linked, but it does not work for me now. We have to wait for a fix :/

afp316, 2023-04-06:
Thank you @cadriel !  That did the trick for me.

afp316, 2023-04-15:
signed off complete based on the @cadriel fix that works for me.

closed 2023-04-15 by afp316

CappyT, 2023-08-17:
Any updates on this? IAQ would be cool to have. The bsec2 lib should be there, I don't know what is blocking the functionality right now

Thanks

Schluggi, 2023-08-19:
> Any updates on this? IAQ would be cool to have. The bsec2 lib should be there, I don't know what is blocking the functionality right now

Yes. This would be nice. The problem is Bosch uses a library called "BSEC" which calculates the IAQ value. BSEC1 is not compatible with any ESP32. Unfortunately is this the version in the current ESPHome version. 

Bosch released BSEC2 some time ago but no one has implemented this into ESPHome yet. Therefore is this (non working) [pull request](https://github.com/esphome/esphome/pull/4585).

There is also a time you could use [this modified fork](https://github.com/Schluggi/esphome/tree/bme68x_bsec) by @neffs to make it work (which uses bsec version 2). But ESPHome had a breaking change so this does not work anymore.

You could try using an ESP8266, since BSEC1 should work there or to downgrade ESPHome to (September 2022 (i think?)) and use my modified fork. But there is a chance other modules uses newer features that were not available at this time.

Since I can't code C++ that well, I'm unable to fix it myself. If you can... I would be happy :)

florianmulatz, 2023-10-13:
Is there a way to include this one as temporary fix?

https://github.com/esphome/esphome/pull/4585

Schluggi, 2023-10-13:
> Is there a way to include this one as temporary fix?
> 
> [esphome/esphome#4585](https://github.com/esphome/esphome/pull/4585)

You are fast.. but I'm as well ;)

Yes, I just created PR #122. But this PR has some breaking changes.
You can try out by editing your packages:

```yaml
packages:
  remote_package:
    url: https://github.com/schluggi/AIOsense
    ref: bsec_is_back # << changed from main
    refresh: 1d
    files:
      # [...] other files
      # - esphome/packages/sensors/bme280.yaml  # comment this line 
      # - esphome/packages/sensors/bme680.yaml  # comment this line 
      - esphome/packages/sensors/bme68x_bsec.yaml # add this line
      # [...] other files
```
Please notice the changed and removed substitutions:
`bmex80_address` -> `bme_address`
`bmex80_update_interval` -> `bme_update_interval`
`bme680_bsec_sample_rate` -> gone. not part of the quickfix
`bme680_bsec_temperature_offset` -> gone. not part of the quickfix

florianmulatz, 2023-10-15:
compiles perfectly :) aaaand -->

```
[09:11:17][C][bme68x_bsec.sensor:134]: BME68X via BSEC:
[09:11:17][C][bme68x_bsec.sensor:139]:   BSEC Version: 2.2.0.0
[09:11:17][C][bme68x_bsec.sensor:141]:   Address: 0x77
```
```
[09:11:17][C][bme68x_bsec.sensor:159]:   IAQ 'IAQ'
[09:11:17][C][bme68x_bsec.sensor:159]:     Device Class: 'aqi'
[09:11:17][C][bme68x_bsec.sensor:159]:     State Class: 'measurement'
[09:11:17][C][bme68x_bsec.sensor:159]:     Unit of Measurement: 'IAQ'
[09:11:17][C][bme68x_bsec.sensor:159]:     Accuracy Decimals: 0
[09:11:17][C][bme68x_bsec.sensor:159]:     Icon: 'mdi:gauge'
[09:11:17][C][bme68x_bsec.sensor:162]:   CO2 Equivalent 'CO2 Equivalent'
[09:11:17][C][bme68x_bsec.sensor:162]:     Device Class: 'carbon_dioxide'
[09:11:17][C][bme68x_bsec.sensor:162]:     State Class: 'measurement'
[09:11:17][C][bme68x_bsec.sensor:162]:     Unit of Measurement: 'ppm'
[09:11:17][C][bme68x_bsec.sensor:162]:     Accuracy Decimals: 1
[09:11:17][C][bme68x_bsec.sensor:162]:     Icon: 'mdi:test-tube'
[09:11:17][C][bme68x_bsec.sensor:163]:   Breath VOC Equivalent 'Breath VOC Equivalent'
[09:11:17][C][bme68x_bsec.sensor:163]:     Device Class: 'volatile_organic_compounds'
[09:11:17][C][bme68x_bsec.sensor:163]:     State Class: 'measurement'
[09:11:17][C][bme68x_bsec.sensor:163]:     Unit of Measurement: 'ppm'
[09:11:17][C][bme68x_bsec.sensor:163]:     Accuracy Decimals: 1
[09:11:17][C][bme68x_bsec.sensor:163]:     Icon: 'mdi:test-tube'
```

Schluggi, 2023-10-16:
> but gas_resistance sensor is gone now.

True. But this was also with the old BSEC the case. Did you really need this value? I have no clue what I should do with this value tbh

florianmulatz, 2023-10-17:
> Did you really need this value?

 Nope to be honest :) Just mentioning.

Cheers