Schluggi/AIOsense #61: 100nF Capacitor information

Triskae, 2023-03-30:
**Category**
- [ ] Some docs are wrong or outdated
- [X] I miss something the docs

**Description**
I realized that I forgot to include the LCSC number for the capacitor parts. While searching for the missing components on the LCSC website, I noticed that there are various versions available for this part. As my electronics knowledge is a little rusty, do you have any additional information about this part?

I assume that it is a Multilayer Ceramic Capacitor, but I'm not sure whether it is leaded or non-leaded. Also, what voltage rating should I use? On the LCSC website, I can see that there are several options available for the same capacitance - 50V, 100V, and 250V. Can you advise me on which voltage rating to choose?

I will update the documentation accordingly OF😄

Schluggi, 2023-03-31:
Reasonable question. This will be part of the BOM (#48).

The capacitor is only to improve readings. It should also work fine without (in most cases).

Leaded or non-leaded does not matter. If you care about the environment chose non-leaded. We use only voltages of max. 5V so anything above is fine. Yes "Multilayer Ceramic Capacitor" is correct. 

I'll add a link later on.

closed 2023-03-31 by Schluggi

Triskae, 2023-04-01:
Thanks a lot for your very clear answer ! 😄