Schluggi/AIOsense #78: PIR not working on ESP8266

CappyT, 2023-05-16:
**Category**
- [?] Hardware / PCB
- [X ] Software / ESPHome

**Version of ESPHome / AIOsense**
2.0.0 board

**Describe the bug**
The esp8266 (D1 Mini) PIR pins are configured incorrectly as reported by esphome, GPIO13 does not have an internal pulldown

```
Failed config

binary_sensor.gpio: [source /config/esphome/.esphome/packages/6d7fe4d6/esphome/packages/sensors/pir.yaml:6]
  platform: gpio
  name: PIR
  id: motion_pir
  pin: 
    number: GPIO13
    
    Only GPIO16 supports pulldown.
    mode: INPUT_PULLDOWN
  device_class: motion
  filters: 
    - delayed_on: 0s
    - delayed_off: 1s
```

I managed to fix 2 other small bugs for the esp8266 (D1 Mini), I don't know if i should send a PR.
- ESP8266 does not have a CPU Temp sensor: https://github.com/CappyT/AIOsense/commit/61ed64bc3006f23a878852ec2ce1a7103066be0a
- ESP8266 led is inverted: https://github.com/CappyT/AIOsense/commit/4c86ee17a23e1bcfa27a9f8556a6ca67bb76b7f6

How can I fix the PIR sensor? I don't know much about electronics, I just know enough about ESPHome

Schluggi, 2023-05-16:
You're welcome :)

ESP8266 are not in use at my flat, so I haven't test it yet. I've fixed the led and CPU Temp sensor based on your changes :)

I've also fixed the PIR ([9111b0f](https://github.com/Schluggi/AIOsense/commit/9111b0faa3e6b22412fb30729240165ae8f79c65)). A pulldown is not really necessary. It's only there because someone has problems with there PIR ([#52](https://github.com/Schluggi/AIOsense/issues/52#issuecomment-1488085553)).

You can try all fixes by changing the `refresh` interval to `1s` and recompile.

```yaml
packages:
  remote_package:
    url: https://github.com/schluggi/AIOsense
    ref: main
    refresh: 1d  # <--- change it to "1s"
```

CappyT, 2023-05-16:
> You're welcome :)
>
> ESP8266 are not in use at my flat, so I haven't test it yet. I've fixed the led and CPU Temp sensor based on your changes :)
>
> I've also fixed the PIR [...]

Thank you for the quick fix, however the PIR stays always on... this is my main file:

`substitutions:
  name: "aiosense-gabry"  # lowercase only, have to be unique
  friendly_name: "AIOsense Camera Gabry"  # device name and entity prefix
  version: "2.0.0"  # your AIOsense version

  bh1750_update_interval: 5s

  bmex80_address: "0x76"  # for some modules you have to change it to 0x77
  bmex80_offset_humidity: "0"
  bmex80_offset_temperature: "0"
  bmex80_update_interval: 30s

  # bme680_bsec_sample_rate: lp # lp -> 3s, ulp ->5min
  # bme680_bsec_temperature_offset: "0"

  # ld2410_max_distance_move: 4.5m
  # ld2410_max_distance_still: 4.5m
  # ld2410_timeout: 5s

  # pir_delayed_off: 1s
  # pir_delayed_on: 0s
  pir_pin_mode: "INPUT"

packages:
  remote_package:
    url: https://github.com/schluggi/AIOsense
    ref: main
    refresh: 1s
    files:
      - esphome/packages/config/base.yaml

      # --- select your board ---
      # - esphome/packages/config/esp32-c3-mini.yaml
      # - esphome/packages/config/esp32-d1-mini.yaml
      # - esphome/packages/config/esp32-s2-mini.yaml
      - esphome/packages/config/esp8266-d1-mini.yaml

      # --- select your sensors ---
      - esphome/packages/sensors/bh1750.yaml
      # - esphome/packages/sensors/buzzer.yaml  # AIOsense >= 2.1.x only
      - esphome/packages/sensors/bme280.yaml  # enable for bme280
      # - esphome/packages/sensors/bme680.yaml  # disable for bme280
      # - esphome/packages/sensors/bme680_bsec.yaml  # does not work yet
      - esphome/packages/sensors/occupancy.yaml  # only if pir & mmWave are on
      - esphome/packages/sensors/pir.yaml
      # - esphome/packages/sensors/rgb_led.yaml  # esp32-c3-mini only
      - esphome/packages/sensors/led.yaml  # d1-mini only

      # --- select your radar sensor ---
      # - esphome/packages/sensors/ld2410.yaml
      - esphome/packages/sensors/sen0395.yaml

# Enable Home Assistant API
api:
  encryption:
    key: "ih3kU+z9htu/Toh8KE9r7vrxQd3+faKuyeLwfPiP6MQ="

ota:
  password: !secret password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # manual_ip:
  #   static_ip: 192.168.1.111
  #   subnet: 255.255.255.0
  #   gateway: 192.168.1.1

# bluetooth_proxy:`

CappyT, 2023-05-17:
After some digging, the problem boils down to the pulldown input. It needs to be there to prevent "floating" (idk if it's the correct term in electronics). My board has this kind of issue.

I'd leave this open (feel free to close it if you want), but I would try to put a physical resistor in the board for ESP8266 compatibility, I'll let you know how it will go.

Schluggi, 2023-07-26:
Closing because of inactivity

closed 2023-07-26 by Schluggi