Schluggi/AIOsense #95: esphome not compliling

Ghome79, 2023-07-21:
**Category**
- [ ] Hardware / PCB
- [ X] Software / ESPHome

**Version of ESPHome / AIOsense**
...

**Describe the bug**

it wont compile with s2 mini or c3 mini

```
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/s2-all.yaml...
Failed config

switch.template: [source /config/esphome/.esphome/packages/f7a0b76d/esphome/packages/sensors/sen0395.yaml:17]
  platform: template
  name: mmWave Sensor
  id: mmwave_sensor
  entity_category: config
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  assumed_state: True
  turn_on_action: 
    - uart.write: sensorStart
    - delay: 1s
  turn_off_action: 
    - uart.write: sensorStop
    - delay: 1s
switch.template: [source /config/esphome/.esphome/packages/f7a0b76d/esphome/packages/sensors/sen0395.yaml:31]
  platform: template
  name: mmWave LED
  id: mmwave_led
  entity_category: config
  optimistic: True
  
  The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead.
  restore_state: True
  assumed_state: True
  turn_on_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setLedMode 1 0
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
  turn_off_action: 
    - switch.turn_off: mmwave_sensor
    - delay: 1s
    - uart.write: setLedMode 1 1
    - delay: 1s
    - uart.write: saveConfig
    - delay: 3s
    - switch.turn_on: mmwave_sensor
```

Schluggi, 2023-07-21:
Should be fixed (#96). Please confirm :)

closed 2023-07-21 by Schluggi

afp316, 2023-07-21:
I was having this issue last night.  I just performed an update with the new config.  I can confirm it works.

Ghome79, 2023-07-22:
yes its working now. 

i wanted to take the time and thank you for your time and effort.
 i also found another bug that i manage to fix myself.

c3 mini with ld2410 i had to reverse the tx rx pin to get the reading. 
I made the changes on the esp32-c3-mini.yaml file,  not sure if its a cpu board related or just the  ld2410.yaml file.

not sure if you wanted a new post for this

Schluggi, 2023-07-24:
> i wanted to take the time and thank you for your time and effort.
no problem :)

> c3 mini with ld2410 i had to reverse the tx rx pin to get the reading.
Looks like the same as #97. I'll cover that soon :)

commit abed4482 (2023-07-21): hanging restore_state to restore_mode  (fixes #95)