Schluggi/AIOsense #107: exclude regions/ create zones

HiTilt-FPV, 2023-08-24:
**Category**
- Software / ESPHome

**Whats your idea?**
Adding a feature that allows to exclude specific regions in the fov of the mmwave sensor to be prohibited from occupation detection.

Example:
If AIOsense is mounted in the bedroom, it would be useful to remove the bed from occupation detection, so the light in the room will turn off if i lie in bed.
With zones, it could turn on the bedside lamp and turn off the ceiling lamp.

The aquara FP2 sensor for example can do something like that, but a little more advanced.

Schluggi, 2023-09-11:
Hi @HiTilt-FPV,

thank you for your input. You can try #112 which introduced up to 4 distance zones. Works ok-ish in my tests. 

Zoning like the FP2 is really advanced and I don't know any mmWave sensor we could support to add this feature. If you know/find one, please let me know :)

HiTilt-FPV, 2023-09-19:
Hi @Schluggi,

as you suggested, i installed the version with distance zones yesterday. But i could not figure out what they do exactly! The new version gives up to 8 slider bars for configuring start and end distances for up to four zones.  But where do i get the output which of the four zones is actually occupied?
I can only activate the following sensors:
- 4x Target (binary sensors which seem to detect different persons)
- 4x Target distance (gives the distance measured to one of the targets)
- 4x Target SNR (target signal to noise... did not test it, but since it's SNR i bet it shows how "good/strong" the targed is detected or visible for the sensor)
- Distance Zone Status (this only tells me if the configured zones were successfully sent to the sensor. i do not get anything else from it.)

florianmulatz, 2023-11-15:
Hi,

ESPHome 2023.11.0 has been released today which includes direct support for sen0395 and also includes section configuration.

[ESPHome 2023.11.0 - Changelog](https://github.com/esphome/esphome/releases/tag/2023.11.0)

Cheers

Schluggi, 2023-11-15:
Thank you @florianmulatz for the update info. I'll add this soon. I just thinking, what would be the best way to implement...

Schluggi, 2023-11-18:
I'll close this issue since we using the official module in #123.

commit 4d426dfe (2023-11-18): adding distance segment zones (closes #107)
closed 2023-11-18 by Schluggi