Schluggi/AIOsense #113: sorry noob question

jangshik, 2023-09-12:
**Category**
- [v ] Hardware / PCB
- [ ] Software / ESPHome

**Whats your idea?**
...
691210910002 << what is this part used for?
I am trying to build the sensor with these components.

ESP32-C3 mini (PCB)
BME280(PCB)
HLK-LD2410C (PCB)
GY-302-BH1750 (PCB)
im going to skip pir sensor, and i  am in middle of ordering other parts. i was wondering what 691210910002 terminal block 2pos is intended for.

Thank you in advance.

jangshik, 2023-09-12:
[image]
im guessing its for this place, but is it required for esp32 c3 mini lolin?

Schluggi, 2023-09-12:
Hi,

the terminal block is optional to power up the board. It fits under the C3 mini but you can also power the board the via USB-C connector of the C3.

Does this answer your question? :)

jangshik, 2023-09-13:
thank you it solved my big question. 
i made the order from AIOsense-v2.1.0.zip file so i think i should be fine.
thank you for this great project!

closed 2023-09-13 by Schluggi