Schluggi/AIOsense #114: More advanced Installation Mehtod for AIOsense

HiTilt-FPV, 2023-09-18:
**Categories:**
- HW: Case
- Mounting/Installation technique or method

**Description:**
I want to submit the installation technique i used to bring AIOsense devices to my household because i think it blends in very good with the surroundings.

**Method:**
**1.** I took the .stl file for the AIOsense stand and created an easy M3 Mount out of it. The advantage of this is that it is usable with typical M3 camera statives and other M3 mounting rigs! For example, i used this aluminum wall mounting rig for security cameras:

[image]
(source: [https://www.amazon.de/dp/B0B5Y4Z7KY?psc=1&ref=ppx_yo2ov_dt_b_product_details](url))

You can find the M3 Mount i designed and used together with some imagery showing an example installation from my home in the attached .zip file!

**2.** I also thought of a method to install AIOsense devices in a way that hides the cables neccessary for the voltage supply of the device. The final Installation (also using the M3 mount from 1.) looks like this:

[image]

This is what you need:

- XH-2A port and XH-2Y plug with crimp equipment (example: [https://www.amazon.de/Crimpwerkzeug-Set-Dechengbao-Dupont-Ratschen-Crimpzangen-Set-Steckern/dp/B09SWXF9Q6/ref=sr_1_29_sspa?keywords=XH+plug&qid=1695031004&sr=8-29-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9tdGY&psc=1](url)
- some white wire with dc positive and negative (example: [https://www.amazon.de/gp/product/B081QLJHX5/ref=ewc_pr_img_5?smid=A18X01B3A742H7&psc=1](url))
- a PCB (example: [https://www.amazon.de/Lochrasterplatte-Leiterplatte-Lochrasterplatine-Leiterplatten-Prototyping/dp/B08YXKRJY1/ref=sr_1_6?crid=28MDC8YAS3TG9&keywords=lochrasterplatte&qid=1695031078&sprefix=lochrast%2Caps%2C90&sr=8-6](url)
- A small DCDC dropdown voltage regulator (example: [https://www.amazon.de/gp/product/B0BN1N88F6/ref=ox_sc_act_title_6?smid=A1A7E5ILEFA1R3&psc=1](url))
- copper tape, 6mm wide (example: [https://www.amazon.de/dp/B09C1WPMZK?psc=1&ref=ppx_yo2ov_dt_b_product_details](url))
- mounting tape (example: [https://www.amazon.de/tesa-Tesa-Schaumstoff-Montageband-5X18-55714-0000-03/dp/B07F46SMV6/ref=sr_1_24?crid=LV50459VL50W&keywords=tesa+schaumstoff+klebeband&qid=1695033240&sprefix=tesa+schaum%2Caps%2C89&sr=8-24](url))
- some masking tape (example: [https://www.amazon.de/tesa-Malerband-einfache-Malerarbeiten-50mm/dp/B0014EFTWK/ref=sr_1_6?keywords=kreppband&qid=1695031487&sr=8-6&th=1](url))
- You also need stuff to solder. Addtitionally, depending if the coppertape you use has an insulating adhesive (like the one i used and linked), you may need to solder the coppertape also!

1. You need to add the DCDC dropdown converters to your AIOsense. Glue it on the AIOsence PCB using the mounting tape. The general Idea is to use an 230v AC to 12v/24v DC converter from any smart home enabled LED lamp you want to install on the ceiling as the power supply for the AIOsense later on. This requires you to permanently power the ACDC powersupply of the LED you want to use, otherwise AIOsense would loose its powersource! Do not forget to set the DCDC converter to 5v output before glueing it into the AIOsense!!!
2. Solder the cable with a XH-2Y plug crimped to the 5v DCDC converter (i cut the cables to about 30cm but 20cm should be sufficient too). Since the case has no hole in it to fit the plug, you need to put the cable through one of the ventilation holes BEFORE soldering it to the 5v converter! Depending on components, the AIOsense may look something like this:

[image]

3. print the coppertape connectorbox and cut the pcb into pieces (about 18mm x 10mm, doesn't have to be perfect) that fit into the connectorbox. glue coppertape around the two ends of the pcb pieces and solder the XH-2A port to it. The poles of the port have to contact the copper tape.

[image]

[image]

4. glue the finished pcb's to the wall/ceiling/etc. close to where you want to install the AIOsense and another pcb where you put your voltagesource/LED ceilinglight.
5. connect the two pcb's electrically using the coppertape (watch out that you get the positive/negative polarity right!!!). depending on your coppertape, you may need to solder a bit! Thest the electrical connections afterwards! (i used a 12v acdc converter and an rgb led strip i had lying around to test if everything worked)
6. tape some masking tape above the coppertape wires (for physical protection, and you can color it using the exact same color from your wall/ceiling (for a mostly invisible installation). Also, you can put wallpaper over it later for a completely invisible installation! If you followed these steps corectly, your installation (befor coloring or putting wallpaper on) should look something like this:

[image]

A finished Installation looks like this (well, the lamp construction is not so final yet... ;-)):

[image]

More pictures in the attached .zip!

[aiosense_ceilingmount.zip](https://github.com/Schluggi/AIOsense/files/12647135/aiosense_ceilingmount.zip)

Schluggi, 2023-10-14:
Looks fancy and a bit over-engineered. I'll pin this issue so it's more likely to be found. Thank you for sharing :)

closed 2023-10-14 by Schluggi

florianmulatz, 2023-10-31:
Those Camera -Holders are much cheaper at Temu -->

[TEMU](https://www.temu.com/at/2-wandhalterungen-kompatibel-mit-kreisformigen-klebe-kameras-und-kreisformigen-innenkameras-verstellbare-montagehalterungen-geeignet-fur-arlo-arlo-pro-2-3-arlo-ultra-cctv-kameras-und-kompatible-kameras-mit-1-4-schraubenkopfen-g-601099528076000.html?top_gallery_url=https%3A%2F%2Fimg.kwcdn.com%2Fproduct%2FFancyalgo%2FVirtualModelMatting%2F817b9c626fc9b23038a5e2065cd1ab1b.jpg&spec_gallery_id=4039992019&refer_page_sn=10009&refer_source=0&freesia_scene=2&_oak_freesia_scene=2&_oak_rec_ext_1=NTE4&search_key=kamera%20wandhalterung&refer_page_el_sn=200049&_x_sessn_id=03b84ysxq3&refer_page_name=search_result&refer_page_id=10009_1698738875237_q8946j45mf)