Schluggi/AIOsense #121: Voice Assist and Wakeonword with INMP441

s-gordon, 2023-10-13:
**Category**
- [ ] Hardware / PCB
- [x] Software / ESPHome

Year of the Voice Chapter 4 introduces wakeonword. It would be great to take advantage of this with the INMP441 microphone in this project.
...

I have an INMP441 connected to a C3 Mini.

This is what I have so far as config as a starting point. Everything looks reasonable to my eyes, but I haven't been able to get this working yet.

```{yaml}
i2s_audio:
  id: i2s_in
  # i2s_lrclk_pin: GPIO5  # WS pin D1 Mini
  # i2s_bclk_pin: GPIO18  # SCK pin D1 Mini
  i2s_lrclk_pin: A5  # WS pin C3 Mini
  i2s_bclk_pin: A1  # SCK pin C3 Mini

microphone:
  - platform: i2s_audio
    i2s_din_pin: A0 # SD pin
    adc_type: external
    pdm: false
    id: mic_i2s
    i2s_audio_id: i2s_in

voice_assistant:
  microphone: mic_i2s
  on_start:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        effect: none
  on_end:
    - delay: 1s
    - light.turn_off: led
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        effect: none
    - delay: 1s
    - light.turn_off: led
```

Schluggi, 2023-10-13:
Hi, we actually never tested the microphone. We add this for "future needs". But since the future is now I will tinker around with this in a few days.

If you are on v2.1.0 you can not use the microphone and the buzzer parallel. Since D6 is used for buzzer and microphone (we were out of pins).

[image]

Schluggi, 2023-10-22:
My working configuration for wake word detection. No problems with this. But make sure you have disconnect the beeper first. This issue will be addressed in hardware version 3.

I think you problem was the missing `start_continuous` part and the from microphone channel (default is `right` but since the PCB set L/R to GND, the channel is now `left`).

My config for ESP32-C3 mini (PCB version 2.1.0):
```yaml
i2s_audio:
  id: i2s_in
  i2s_lrclk_pin: A5  # WS
  i2s_bclk_pin: A1  # SCK

microphone:
  - platform: i2s_audio
    id: mic_i2s
    adc_type: external
    i2s_din_pin: A0  # SD
    i2s_audio_id: i2s_in
    pdm: false

voice_assistant:
  id: va
  microphone: mic_i2s
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  on_listening:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 0%
        red: 0%
        green: 100%
        brightness: 100%
        effect: pulse
  on_end:
    - delay: 100ms
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }

script:
  - id: reset_led
    then:
      - if:
          condition:
            - switch.is_on: use_wake_word
            - switch.is_on: listen_light
          then:
            - light.turn_on:
                id: led
                blue: 100%
                red: 100%
                green: 0%
                brightness: 100%
                effect: none
          else:
            - light.turn_off: led

switch:
  - platform: template
    name: "VA Use Wake Word"
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
      - script.execute: reset_led
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - script.execute: reset_led

  - platform: template
    name: "VA Listen Light"
    id: listen_light
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - script.execute: reset_led
    on_turn_off:
      - script.execute: reset_led

esphome:
  min_version: 2023.10.0
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

```

closed 2023-10-22 by Schluggi

Ghome79, 2023-10-30:
well hello, I've tested the above config with my c3 mini and without buzzer and i cant get my microphone to work. i have installed and setup openwakeword. piper, and whisper. i also tried 4 different microphones. i believe i cant get it to listen cause my rbg led stay purple

Ghome79, 2023-11-01:
Thx for fast reply 
ill let you know where I'm at with 1 full day of testing. 

All my testing is done with all sensor disabled, 
I know my pipeline assist works via web 

esp-s2 sorta works cuz I need to shout into mic to enable it. very frustrating
esp-c3 doesn't work even if I shout with the same mic  and setting