Schluggi/AIOsense #125: change to ESPHome integrated sen0935 feature

florianmulatz, 2023-11-15:
**Category**
- [ ] Hardware / PCB
- [X] Software / ESPHome

**What would you like to improve?**
I would suggest to use the "own" integrated ESPHome feature for driving the sen0395 sensor so no specific effort is needed to take care on an own integration.

[Changelog](https://github.com/esphome/esphome/releases/tag/2023.11.0)

Cheers

Schluggi, 2023-11-15:
Hi @florianmulatz, thank you for noticing. I just replaced it. Feel free to give feedback in #123.

closed 2023-11-15 by Schluggi

florianmulatz, 2023-11-21:
> Hi @florianmulatz, thank you for noticing. I just replaced it. Feel free to give feedback in #123.

Working perfectly! - Automatically created configuration for 3 individual segments.

commit b44154ef (2023-11-15): replacing the sen0395 component the new official one (closes #125)