Schluggi/AIOsense #190: ESP32-S3 Support with microWakeWord

AnthonyPluth, 2024-03-13:
**Category**
- [x] Hardware / PCB
- [x] Software / ESPHome

**Whats your idea?**
With ESPHome now supporting on-device wake word detection (https://www.home-assistant.io/blog/2024/02/21/voice-chapter-6/), it would be awesome to add support for ESP32-S3 with microWakeWord onboard. Happy to collaborate on this work, as well.

lukas-holzner, 2024-06-26:
[image]

microWakeWord needs the PSRAM which needs GPIO Pins 33 to 37. Unfortunatly we are currently using some of those pins for the microphone and speakers so a redesign of the pcb would be needed.

AnthonyPluth, 2024-06-26:
Are there free pins on the ESP32-S3 that could be utilized instead (with PCB redesign)?

s-gordon, 2025-01-27:
@AnthonyPluth I'm looking at redesigning the PCB to free up those PSRAM pins. I've done a little probing over the last few days to see what pins might make sense. This is what I am thinking (using the [S3 Mini layout](https://www.wemos.cc/en/latest/s3/s3_mini.html)). Thoughts?

- SCL -> GPIO16
- SDA -> GPIO18
- I2S data out (speaker): GPIO17
- I2S data in (mic): GPIO21
- I2S LRCLK: GPIO12

AnthonyPluth, 2025-01-27:
> @AnthonyPluth I'm looking at redesigning the PCB to free up those PSRAM pins. I've done a little probing over the last few days to see what pins might make sense. This is what I am thinking (using [...]

I'm probably the wrong person to ask for this. However, in my experience, the S3 Mini seems to be a bit light on RAM. IMO, it's worth upgrading to an ESP32-S3 N16R8V. Unfortunately, that requires increasing the size of the board, but it would allow you to place the speaker on the front of the board instead of the back.

Schluggi, 2025-02-02:
No redesign needed. It works "just fine" with the current pinout.

But the PSRAM is indeed to small and fully filled by just the wake word detection. Therefore I just contacted Lolin (the vendor of the S3-mini). I'll be back after getting response.

Moreover `micro_wake_word` uses the newer `esp-idf` framework instead of the current used arduino framework. Most components could be migrated, but [BSEC would not be available for now.](https://github.com/esphome/feature-requests/issues/2431).